sim-/tgy #66: [idea] rotorSENSE

grawlinson, 2015-01-14:
Just wondering if it would be possibel to implement immersionRC's [rotorSENSE](http://www.immersionrc.com/introducing-rotorsense/) to the SimonK firmware?

sim-, 2015-01-14:
Hello! I've actually done this a few years ago in some testing firmware, and I've been wanting to make it more of a fully-driven menu selection system, but I am not sure how best to activate it. For example, we don't; want an accidental reset while the motor is spinning to accidentally reset the rotation direction...though in theory it would still be rotating the right way in most cases. :)

Perhaps it could just detect motor movement as long as a high pulse is received, where it would normally start calibration. Forcing the pin high is currently used by the boot loader, however. Hmm. Ideas?

grawlinson, 2015-01-15:
Would it be feasible to detect an unusual pattern of some sort such as:

1. Movement in the desired direction.
2. Zero movement *or* movement in the opposite direction.
3. Movement in the desired direction.

I can understand not wanting to unintentionally trigger this, so what other possible safeguards are there without making the process convoluted?

sim-, 2015-01-15:
Well, it takes no more power to watch for it at any time, so perhaps just movement in the desired direction for long enough to be more than just a bump, stop, esc beep, then rotate again in the same direction. This wouldn't happen naturally and wouldn't change the setting if you merely bump it (though in theory you could bump it twice) and isn't much harder.

Hmm, it seems that this is exactly what rotorSENSE does! Yay for obvious minimal approaches. ;) It also seems to say patent not pending, so perhaps I'll do that. I would also like to come up with some other dance to clear (reset) calibration. Perhaps holding the input pulse high for >10s or something.

The original idea also was to make the motor act as an input knob with magnetic "detents". This could be used to navigate menus easily, though it would have to keep pulsing the motor once this mode is active in order to detect small movements. The problem I had with this idea was coming up with something easy when there is no feedback other than beeps and feeling. I hesitate to add the direction sense feature really only if it steps on this other idea.