sim-/tgy #68: Building with AVRA

aliktab, 2015-02-02:
Hi there,

I spend a lot of time trying to build this project with AVRA (v1.3.0) on my Mac but without success.
It crashes on the line like this one:
   orig_osccal:	.byte	1	; original OSCCAL value
with 
  Pass 1...
  m8def.inc(44) : PRAGMA directives currently ignored
  m8def.inc(48) : PRAGMA directives currently ignored
  m8def.inc(53) : PRAGMA directives currently ignored
  m8def.inc(690) : PRAGMA directives currently ignored
  m8def.inc(691) : PRAGMA directives currently ignored
  m8def.inc(692) : PRAGMA directives currently ignored
  m8def.inc(693) : PRAGMA directives currently ignored
  m8def.inc(734) : PRAGMA directives currently ignored
  Abort trap: 6

Any suggestions? Does anybody build it on Mac?

sim-, 2015-02-17:
Hello! This sounds like an avra bug. https://github.com/sim-/tgy/pull/67 may work for you, but it's still an avra bug that is causing an "Abort trap: 6" at the bottom there. The PRAGMA directives ignored output are just warnings. I'm shipping the files with them anyway because then it also works with AVR's assembler. It would certainly be nice to get to the bottom of this..

aliktab, 2015-02-17:
Hello.

Looks like it works thanks. But it's necessary to fix all lines including data definition. For example avara crashes on line
	orig_osccal:	.byte	1
and don't crash if
	orig_osccal:	
		.byte	1
are you going to merge such changes to the master?

sparky-p32, 2016-05-17:
Alternately, there is a patch to fix the bug in avra (due to improper use of strcpy() for overlapping buffers).

See: https://sourceforge.net/p/avra/patches/16/

adammck, 2017-03-25:
This issue was still present a few days ago, but [is now patched](https://github.com/Homebrew/homebrew-core/pull/11285) in the latest AVRA package in homebrew. I haven't managed to get it merged upstream yet.

closed 2021-09-14 by aliktab