sim-/tgy #69: Low THR while fast copter tilt

icsss, 2015-02-10:
I'm having this odd issue with naze32 fc and mystery blue series 30A esc + simonk BS V2014 09 30.

If I keep the quad in hand while low throttle (<50%)... and (very) quickly tilt it to one side, the esc fails to correct the attitude - it just keeps the motor at low throttle.

This doesn't happen if I slowly tilt the quad. Also, hover and slow flying are good, the problem appears while aggressive flying.

I have several ESCs (same brand and specification but different batches), some of which have this problem, some don't. All were flashed with the same simonk version, the only difference is the "reverse" option. I'm not sure but it seam to have this problem only with "normal" simonk, not "reverse".

I some strange way, this looks to be related to environment temperature and seams to emerge more obviously if flying below 0 degrees C.

icsss, 2015-02-10:
PS: the motor PWM showed in GUI look right... over 1800++

icsss, 2015-02-15:
Solved - Twisting the wires and adding a ferrite ring did the job.
More details here: https://github.com/sim-/tgy/issues/50

closed 2015-02-15 by icsss

sim-, 2015-02-17:
Hello! Interesting feedback here. It seems you've now solved the issue, but can you perhaps post a picture on what the setup looked like (preferably before adding a ferrite/twisting)? I've never seen this be a problem except on large configurations (with long arms).

I can't see how it would be related to reverse-or-not, but if it were a temperature-sensitive problem,   those ESCs typically have ceramic resonators making them fairly temperature-stable, but if the fuses are set (for example) to not use the external oscillator, they would become fairly temperature sensitive and that could make a sudden jump to full power fall into the invalid pulse length range if it (appears to be) over 2400µs. However, it should time out and shut off within ~128ms of that.

...but that doesn't seem to be the problem. So, I'm mostly interested in how you wired it to reproduce the noise problem.

icsss, 2015-02-17:
Sure, I will try to post a few pictures later today.

During the trials, I have flashed reverse/normal and concluded that the problem is not related to this.
About the temperature I would say yes, it affects the ESC behavior. For me to replicate the problem, I had to keep the ESC to about 0 degrees Celsius, for about 20 minutes.

Interesting is that only one ESC had this problem, and I assume it is related to the wires position or maybe length? The length is about 20cm and fairly close to the frsky RX.

Pictures later today...

icsss, 2015-02-17:
Here are some high resolution photos with the working setup:
https://drive.google.com/folderview?id=0B-rvKLM4BMPEfjBCMlV4MXdIMU91NGtJUTJrSVQxWHdWRFJ3UDhxYVJsRTh5T0k5ckNCVEE&usp=sharing

In one image, I have tried to draw the cables, since the propulsion and ESC cables are not visible.
I have also 2 FrSky Battery Voltage Sensors, one for the FPV lipo and one the propulsion lipo. There is also a voltage beeper (with lost&found option) that is connected directly to the propulsion cables. During the tests, I have disconnected the beeper and the voltage sensors from the RX, however they remained connected to the lipo.

The difference between the working setup and faulty setup is the twisted ESC cable + ferrite ring. The wire path was roughly the same in both cases. FPV gear was disconnected.

Beeper: http://www.hobbyking.com/hobbyking/store/__60454__Turnigy_3_In_1_Battery_Monitor_Signal_Loss_and_Lost_Airplane_Alarm.html

sim-, 2015-02-18:
This still doesn't really make any sense. Your pictures show pretty short wires, and everything else is high enough frequency that really the only thing that could cause interference is the PWM switching, like per https://github.com/sim-/tgy/issues/50 , but it's weird that you would have that problem only when cold unless it's a connection problem. I think the oscillator offset is too little for it to be a drift problem (though you could check the fuses on that ESC), but is it possible that maybe there was a bad connection before you twisted the wires? Temperature can certainly make things shrink, too...

icsss, 2015-02-18:
I have tested 3 ESCs and all behaved the same way. When replacing the ESC, I kept all old wires and soldered the ESC in position.

Connections were checked thoroughly and wires replaced, but the only thing that made a difference was twisting them. I could replicate the problem each time by (un)twisting the wires.

How can I can check the fuses on that ESC?

Other thing I have tried but didn't solved my problem: 
- swapped motors
- changed cppm/ppm
- flashed several simonk versions
- changing RX and RX antenna position in relation to the ESC wires
- disconnected all other electronics: e.g. FPV gear
- replaced ESC