sim-/tgy #75: Trying to use Oneshot125 results in insufficent throttle control in 3D mode (RC_PULS_REVERSE)

SystemsGuy, 2015-03-29:
Tested using cleanflight 1.8.0 on a Naze32.  Compiling bs_nfet with RC_PULS_REVERSE = 1 results in a clean build and load.  When full throttle is commanded from the FC with Oneshot125 enabled, the motors deliver approximately 1/3 power at full and reverse, and have a huge deadband in the middle.

When disabling Oneshot125 on the FC, the motors respond as expected, with full power at max and full reverse at min.

Manual (sticks) calibration using motor command in cleanflight cli results in the expected sequence, ending in the three beeps at neutral.

Recompiling with RC_PULS_REVERSE = 0 and using Oneshot125 on the FC works as expected, with no power at min, and full power at max.

One of the CF developers did some signal level debugging as documented here : https://github.com/cleanflight/cleanflight/issues/680

sim-, 2015-04-07:
Hello! Hrmm, I hadn't thought about this use-case, so yeah, I'm kind of not surprised. Likely the deadband offsetting is wrong, etc. Will check.

sim-, 2015-05-05:
Want to give master a try?

SystemsGuy, 2015-05-08:
Just pulled and complied - looks great on the logic analyzer, and flys well from the limited testing in the garage!

closed 2015-05-08 by SystemsGuy