sim-/tgy #76: Can I use this firmware with pinmap like this? Which inc file should I use?

anybye, 2015-04-06:
Pinmap like this

mux_c=PD7  
c_comp=PD6
ApFET=PD5
BpFET=PD4
CpFET=PD3
rcp_in=PD2

mux_a=ADC7
mux_b=ADC6
temp_adc=ADC1
accu_adc=ADC0

AnFET=PB2
BnFET=PB1
CnFET=PB0

MCU is MEGA8 with 8Mhz
I really don't know how can I make it wok. Can you hlep me? My English is not very good. Thank you very much.

sim-, 2015-04-07:
Hello! Well, sure. Which ESC is this for? That differs from tgy.inc only with mux_a and mux_b, so to try you could just change those in tgy.inc and it should work.

anybye, 2015-04-07:
Thank you for your reply.This ESC is based on TP-18A but some pins have been changed . I'm not sure if the firmware works. I'll try it later and  let you know the results. 
thank you!

anybye, 2015-04-07:
I have another question. Whether the boot_loader=1 is necessary ? If I don't want it ,what should i do except set boot_loader=0 ?

anybye, 2015-04-07:
Hello sim. I have flashed the hex file into my ESC and i  get only one beep. Is it right? lfuse=a4 hfuse=d9 with bootloader enable.I'm sorry to take up your time and  thank you again.

anybye, 2015-04-07:
never mind.   It works. Thank you so much.

sim-, 2015-04-07:
BOOT_LOADER is if you want to build in the boot loader or not. If 0, the resulting .hex will not have any. BOOT_JUMP sets whether it should jump to one on high input signal. You can use another boot loader, for example, by setting BOOT_LOADER to 0 and BOOT_JUMP to 1. Or, just set both to 0 and use only ISP.

Sounds like you had an input signal problem, but corrected it (inversion?).

Is this going to be a shipping product? Would you like to add support for it to to the tree?

anybye, 2015-04-08:
Hi sim . Thank you for your reply. It just a DIY product acturlly. One of my friends soldering it. The first firmware is tp-18a_ppm_Ro5 but i find it only works @50HZ input.When I conect it to APM&CC3D it didn't work. I have to set APM's output signal to 50HZ for this ESC. So I find your firmware and it's great. This ESC can work with CC3D's 400HZ output pwm now, have been tested.We make this ESC just for fun. Thank you for your help and i learn a lot from it. About the problem i guess it is a connect mistake maybe. Whatever , it works!

closed 2015-04-08 by anybye