sim-/tgy #77: Multistar 4225 390kv motors and Afro 30amp opto esc's can't fix sync issues.

Paulo-uk, 2015-04-09:
Hi Simon, Im using the following setup and having lots of syncing issues - Please help:

Multistar 4225 390kv motors
Afro 30amp opto esc's
6s 5000mah 35c Lipo
12x6 props

Is there a recommended firmware version that will work here or am I doomed?

Thank you in advance.

Paulo

sim-, 2015-05-15:
Hi Paulo,

No, I don't think so. I have similar motors and ESCs and it works for me. Which flight controller are you using, and did you try updating to the latest firmware?

Simon-

Paulo-uk, 2015-05-18:
Hi Simon,

Im using a DJI wookong on the latest firmware…

Regards,

Paul

sim-, 2015-08-09:
Hi, Paulo! Sorry for the delay. If you still have this problem, I'm curious if twisting the servo cable (just twist it around by hand until it begins to buckle) and then re-attaching it to the flight controller makes any difference or not for you.

If you've solved it another way, or if you've given up in the meantime, I'm also interested to hear. Thanks!

Paulo-uk, 2015-08-13:
Hi Simon, Im afraid I had to switch escs. Thanks anyway

sim-, 2015-08-13:
Ok, no problem. Which ones ended up working?