sim-/tgy #78: Problem after flashing ESC Afro 12A Opto

mashu, 2015-04-11:
# How to reproduce
I flashed Afro 12A Opto with latest firmware (master) using RapidFlash - https://chrome.google.com/webstore/detail/rapidflash and using Afro USB programmer over servo wires.
- Enabled COMP_PWM
- MOTOR_ADVANCE set to 15 (medium)

With default motor advance set to 18 problem was even more apparent.

I am using Openpilot Revolution board with Oneshot125.

# Problem
You can hear high pitch vibrations at throttle 80-100%. 
Video link is here https://vimeo.com/124685609

# Solution ?
My questions 
- What are the recommended correctly functioning settings for Afro 12A ?
- Is lowering the MOTOR_ADVANCE a proper solution for my problem ? (What are disadvantages)
- What is the optimal configuration for OneShot125 ?

sim-, 2015-04-13:
Hello! This is actually the result of hitting TIMING_MAX. It works sort of like a bad car engine rev limiter.

If you set TIMING_MAX to something like 0x80 rather than 0xe0, the problem should go away. However, the reason it is set to 0xe0 is because timing starts to lag at faster values than that. Nonetheless, it should still work in your case, but the motor won't quite reach the maximum speed/acceleration that it could if the code (or CPU) were faster.

I've been working on fast-pathing the crazy 24-bit math, but it's a little nutty. The main issue is calculating the timing. I could switch it back to 16-bit, but then things like oneshot125 support would have been even more tricky and the resolution would drop back to half-microsecond. Everything is calculated currently at full 16MHz ticks, which is why it's a bit tight on an 8-bit MCU.

sim-, 2015-05-15:
By the way, could you try changing TIMING_MAX to 0x80 and see if the problem disappears? Just to confirm it is the same issue.

mashu, 2015-05-15:
I wish I could test it, but I flashed different firmware. I may check that in future though, unless someone else can check this faster.

kyau, 2015-05-27:
I was having this problem on my Afro 12A v3 w/ BEC, I tried the TIMING_MAX change like you suggested and this fixed everything now running smooth with COMP_PWM.

quadromania, 2015-07-28:
hello a read your massage about fixing the Afro 12A ESC, i`ve the same problem
 Where i have to change  the ; Timing_Max in cleanflight ore rapidflash
canot find it annywehere

please help, i`m grounded

sim-, 2015-10-01:
Hello! This is fixed in 103edb5a08571cd4b7ccf0be5427eb3065666533, released as 2015-09-12. TIMING_MAX was changed as discussed, but also a fast-path added to make the timing not skew, so it should have increased power at full speed.

closed 2015-10-01 by sim-