sim-/tgy #79: Release TBS Gemini ESC firmware with OneShot125 support

CrazyCoder, 2015-04-12:
Not sure if it's planned to make this firmware available to public, but why not if we already have other TBS specific firmwares in the repo?

Users requested this firmware to be released in TBS Gemini Owners facebook group, Trappy said that it's already available and being tested, however no files for the firmware were published.

In case it will remain closed, could you please at least let us know if this firmware will support COMP_PWM? Thank you!

sim-, 2015-04-17:
Hello! There are no closed forks. Oneshot125 is already there in HEAD and should work for all targets. I need to tag a release still, though.

CrazyCoder, 2015-04-17:
Thanks, I know about OneShot125. The question was which specific firmware goes into TBS Gemini ESC and if it supports COMP_PWM.

CrazyCoder, 2015-04-17:
I'm referring to this post in the TBS Gemini facebook group, one and a half months ago it was promised to give us the new firmware ASAP, but no news since then:

[image]

CrazyCoder, 2015-04-17:
I've guessed the wiring, was able to read the original firmware, but can't identify it reliably (compared to all available SimonK firmware hex versions with default settings):

[image]

[image]

CrazyCoder, 2015-04-17:
[Original firmware hex](http://pastebin.com/DWzQBXeX) (looks partial, probably because it's locked with the fuses?)

sim-, 2015-04-17:
Hmm, is that comparison match % chart part of BLHeliSuite? I should check that out...

That .hex isn't partial, it's just skipping over the gap before the boot loader. Lock fuses just give you address back as data instead.

Anyway, with some avr-diff scripting and archives I can find that it matches 2014-09-30 tbs target with COMP_PWM=1, DEAD_LOW_NS=500, DEAD_HIGH_NS=500, MOTOR_REVERSE=1, and RC_CALIBRATION=0. That should get you a matching file. I'll confirm which variants they are using and flatten it back to tbs.inc which should reduce confusion in the future.

CrazyCoder, 2015-04-17:
Thanks! Yes, this comparison match chart is a part of BLHeli Suite, quite handy to detect ESCs firmware (BESC Tools | Identify BESC).

So, the ports configuration is the same as in tbs.inc and it should work if I build the master version with the settings suggested above to get OneShot125 support?

CrazyCoder, 2015-04-17:
Also, Trappy said that their firmware is customized to stop motors from spinning in case of a crash (when motor stalls) to save ESCs. Is it true for all tbs.inc ESCs or is there some option one has to enable?

sim-, 2015-04-17:
Yes, master should work. Yes, start power modulation and give-up is there since 2014-09-30, as implemented by 61a017f04e and b25c880c8a.

I've been trying to release a new version but keep wanting to fix #78 first. :)

CrazyCoder, 2015-04-17:
Great! Thanks a lot for your help, I'll try to flash the new firmware once I figure out the IO board layout on Gemini.

I guess MOTOR_ADVANCE is the default (18) in the stock firmware that I've uploaded?

sim-, 2015-04-17:
Yup. You can change it if you'd like..higher will make it go a little faster at the top end, and accelerate faster, but whine a bit more with COMP_PWM and increase diode conduction. Lower will be the opposite.

CrazyCoder, 2015-04-18:
Flashed the ESCs thanks to this image which helped to understand which IO Board pins correspond to motor outputs:

[image]

With Afro USB Linker, connect to the **first 6 pins on the right side** one by one and flash all the ESCs with the new firmware:

[image]

**Here is the firmware hex that I used**: http://bit.ly/1b67ScZ
It's tbs.hex from master built today with `COMP_PWM=1, DEAD_LOW_NS=500, DEAD_HIGH_NS=500, MOTOR_REVERSE=1, RC_CALIBRATION=0`).

I used [RapidFlash](https://chrome.google.com/webstore/detail/rapidflash/gehadojofkekobiohnefkabgimeniglh?hl=en), **Load Local Firmware**, point to the saved `.hex` file, **Flash Firmware**:

[image]

After flashing the firmware refer to https://github.com/TauLabs/TauLabs/wiki/User-Guide:-OneShot125 for the instructions how to enable OneShot. One has to use the current Tau Labs build from the `next` branch, grab a nightly binary at http://jenkins.taulabs.org . Note that outputs range has to be changed from 1500 to 3000 in GCS configuration.

My settings:

[image]

[image]

I did a quick hover test and it flies great! Time to fly outside. I guess I'm the first outside TBS to fly Gemini with OneShot125, but now everyone can do it!

sim-, 2015-04-22:
So, I'm guessing you didn't do a double blind test to see if it's any better..because you know, if you're blind, you'll probably crash it! ;) Seriously, have you noticed any perceivable difference?

CrazyCoder, 2015-04-22:
Haha, good question. Unfortunately, I had no chance to compare as it was basically a maiden flight in MWRate mode, before I just did quick hovers in stabilized mode.

I believe it should help with the small twitches and make PIDs more forgiving.

Here is the video from the test flight: http://www.youtube.com/watch?v=qtahb5Oo8wA

sambas, 2015-05-15:
Flashed mine with this firmware. Sadly one esc didn't have bootloader in it so had to ISP it. Did some before and after logging with same tuning parameters. Looks better.

stock: https://www.dropbox.com/s/1wiipa9whs52lkw/stock.png?dl=0
vs
oneshot: https://www.dropbox.com/s/77v454dphex04pw/oneshot.png?dl=0

https://www.youtube.com/watch?v=xuOXf0vlSLA

ChrisPrefect, 2015-07-01:
I don't get the oneshot option in the system settings..

I flashed the ESCs and used "taulabs_next_20150701_211246_ca3e5d00f0_win32.exe". 

The firmware of the colibri is 20150701 21:12

I only see PWN and three other options in the dropdown under ChannelType. Where did I go wrong?

CrazyCoder, 2015-07-01:
Please ask TauLabs developers, it was removed/changed for some reason in the most recent nightly firmware. I used 20150415 version where OneShot option was available.

ChrisPrefect, 2015-07-02:
Thanks! But I cant find version 20150415. Is it still the "next" branch? I did not find it on the main repo either.

CrazyCoder, 2015-07-02:
@ChrisPrefect [here is the version that I used](http://www.mediafire.com/download/bqw6j8tkq7r8k7n/taulabs_next_20150415_072002_bd1b9d0a4c_win32.exe).

ChrisPrefect, 2015-07-02:
@CrazyCoder Thank you!!

I read something that they changed the name of the OneShot option to SyncPWM or something like that. But I did not find that in the latest version either.