sim-/tgy #82: license?

richo, 2015-06-14:
Hi! I was looking through the repo, and was curious to know if the code is released under a license?

Apart from Bernard's license in tgy.asm I wasn't able to find one. Thanks!

sim-, 2015-10-10:
Hello! I've not changed Bernard's license, as I understand it is not possible for Germans to transfer copyrights and so doing that would require some coordination, and I'm also not sure which different license would be best. I am not a lawyer. :) Suggestions welcome.

richo, 2015-10-10:
Personally I prefer the MIT license, which I *think* is compatible (but I also am not a lawyer).

lazyzero, 2015-10-12:
I think it is not compatible to most open source licenses. 

Roughly translated version of Bernard’s license.

>1. Because I make the firmware available free of charge, I offer not warranty for the function and are not liable for any damages.
>2. The software is only for private use. I have not checked if it is in conflict with any patents or if there are any other legal conflicts.
>3. Everybody is allowed to change it to his needs. I would like to see that you keep me as a co-author in the documents. Please send me a link to your changed version.
>4. The changed software must stay free, (he means free of charge)
>Who is not okay with this rules is not allowed to use the software.

Best Christian

dragonet80, 2015-10-12:
I think it feets perfectly to CC-BY-SA-NC.

lazyzero, 2015-10-12:
Cite from creative commons FAQ

>Can I apply a Creative Commons license to software?

>We recommend against using Creative Commons licenses for software. Instead, we strongly encourage you to use one of the very good software licenses which are already available. We recommend considering licenses made available by the Free Software Foundation or listed as “open source” by the Open Source Initiative.

dragonet80, 2015-10-12:
Interesring, thanks to point. I always believed CC could also be aplied to
software. In fact, open hardware sometimes requieres software or firmware
to function. I thougt that CC, in that case, was appliable.
In fact, they say it is possible to use it for software, but strongly
encourage you not to do it... that seems vague. Conclusion: try to find
another license, but if you do not, use CC.
GNU GPL v3 is compatible with CC-BY-SA:
http://www.fsf.org/blogs/licensing/creative-commons-by-sa-4-0-declared-one-way-compatible-with-gnu-gpl-version-3
It lacks the Non Comercial part. But the FAQs of GNU says that the license
GPL implies the software to be Free:
http://www.gnu.org/philosophy/free-sw.en.html
So, I assume GPL is compatible with BY-SA-CC, but I can be wrong.

Cite from creative commons FAQ

Can I apply a Creative Commons license to software?

We recommend against using Creative Commons licenses for software. Instead,
we strongly encourage you to use one of the very good software licenses
which are already available. We recommend considering licenses made
available by the Free Software Foundation or listed as “open source” by the
Open Source Initiative.