sim-/tgy #83: ZWT 12a (Blueseries 12a) - "copy" binaries?

swilkens, 2015-07-25:
ESC: ZWT Mantis 12a - same as the blueseries 12a

Issue:
As I can't recall the settings I used to generate the bs_nfet hex for these ESCs, I'm trying to copy the flash memory and eeprom contents off the ESC so I can flash them back to another. Here's what I've done so far.

Using an Arduino mega as the ISP programmer, reading eeprom and flash to file:
1. avrdude.exe -P COM5 -b 19200 -c avrisp -p m8 -v -U flash:r:simonk_flash.hex:i
2. avrdude.exe -P COM5 -b 19200 -c avrisp -p m8 -v -U eeprom:r:simonk_eeprom.hex:i
3. Note the lfuse and hfuse

I then tried to restore these dumps to another ESC of the same model:
1. avrdude.exe -P COM5 -b 19200 -c avrisp -p m8 -v -U flash:w:simonk_flash.hex:i
2. avrdude.exe -P COM5 -b 19200 -c avrisp -p m8 -v -U eeprom:w:simonk_eeprom.hex:i
3. evrdude.exe -P COM5 -b 19200 -c avrisp -p m8 -v -U lfuse:w:0x2e:m -U hfuse:w:0xcf:m

My issue being that I can't get this to work. The flash completes successfully, the data verifies. The flashed-to ESC just isn't functional. What am I missing here?

Apologies if this isn't the proper location to ask these things. I was unsure where else to ask.

swilkens, 2015-07-25:
To follow up, I did an eeprom dump (on the avrdude terminal) on the source ESC and the destination ESC, eeprom content isn't the same. Clearly I'm doing something wrong.

swilkens, 2015-07-26:
And one last follow-up.

It seems the original ZWT mantis (simonk 20130515) has the lockbits set to prevent verification, I assume this also means I can't read the chip. I would have to do a full erase to be able to alter the fuses, thereby losing the contents of the chip?

swilkens, 2015-07-26:
Well sorry for all the spam, closing this up:

The fuses were the cause, they prevented a read-out of the original ESC causing me to flash junk to the target ESC. Manually re-building with AVRA and flashing that to the target ESC solved everything.

Live and learn.

closed 2015-07-26 by swilkens

sim-, 2015-08-24:
Hello! Sorry for not noticing this before. Just some notes.

I have no idea what ZTW used since I've never actually worked with them, but if it's a version with a working boot loader, it's possible that you could read the EEPROM contents from there (with an Afro/Turnigy USB linker or Arduino-ish linker or similar), since I don't think the fuses can prevent EEPROM readout from the boot loader.

This sort of silliness is why Steffen added a fuse check to BLHeli to prevent it from working if the manufacturer applies lock fuses, but in cases like this they'll probably just keep shipping a version from 2013 anyway :), and it can cause other problems (http://owenduffy.net/blog/?p=2388).

Good to hear it's working now at least!