sim-/tgy #84: comp_pwm too weak for active braking

dustin, 2015-08-07:
I've been seeing various reports of comp_pwm being excessively weak as compared to blheli dampened light.  I do not have any first-hand knowledge of the difference, but it's been used as a reason to flee SimonK firmware.

The most complete report I've been able to find is that it has the following undesirable behaviors:

> when throttling down has both weak braking and sometimes ESC shutdown or lag out causing unexpected flips

as well as

> starting from min_throttle a motor will not spin up in time

Both of these conditions were allegedly resolved by installing blheli.

sim-, 2015-09-10:
Hello! This should be fixed by the 2015-09-09 release; specifically, we were not previously mirroring the complementary state following a commutation step. Instead, we were waiting for the next PWM interrupt. This would effectively reduce the braking force proportional to the speed, inversely to the PWM frequency.

BLHeli also has negative dead-time which can help on boards with slow high-side drives, like the BlueSeries and Afros with discrete drive components. I will look at that separately, but it's a bit dangerous to support, and it'd be better to just fix the hardware.

closed 2015-09-10 by sim-

dustin, 2015-09-11:
Very grateful for the work.  Thanks for getting this done.  I'm looking forward to trying it out.