sim-/tgy #85: How to disable timeout?

vishnubob, 2015-08-11:
Hi Simon (et al)!  I love this project, thank you so much for contributing it to the community.  I'm using ESC's in non-RC applications, and your firmware allows me to tweak off-the-shelf ESCs for my applications. Recently, I loaded a Spider 30A with ```tgy``` to spin a hard drive motor for an [HDPOV](http://www.instructables.com/id/Hard-Drive-Persistence-of-Vision-HDPOV/) display.  I'm using an Arduino to control the system, and since I've allocated all the timers for other purposes, I'm controlling the ESC via the UART.  The problem I'm having is that sending the ESC constant refreshes via the UART is introducing jitter errors in my display code.  Therefore, in the case when ```tgy``` doesn't receive an update, I would like to do modify its behavior to simply use the last duty value instead of timing out and disarming the motor.

I've spent a bunch of time going over the code, but I was hoping you could give me a few hints about how I might accomplish this.  I gather ```rc_timeout``` and ```RCP_TOT``` are relevant variables, but I get kind of lost in the larger state machine that comprises the startup and run modes of the firmware.  If you have a few minutes to spare, can you tell me where should I focus my analysis, and what are some design considerations I should take into account?

sim-, 2015-08-11:
Hello! Cool project!! :)

These values and variables have changed a bit over the years as well. I introduced a shorter timeout while running due to some other awful software (i86) and situations (pulling the wire at a bad time) that can do things make a full-power PWM pulse and then stop. When armed, RCP_TOT sets how many timer ticks (65536µs periods,  t1ovfl_int) need to pass before rc_timeout is decremented to 0 and thus power is cut. When rc_timeout is 0, rct_beacon is used to add a ~2 second timer to actually disarm the ESC (to work around crappy radio and other problems that would happen if it just disarmed immediately).

You probably don't really need to touch anything other than to comment out "cpse rc_timeout, ZH" so that wait_commutation always returns rather than bailing. Additionally, if you'd like to make arming easier, etc., you could rip out everything to do with rc_timeout, or just comment out the "dec rc_timeout" in t1ovfl_int.

Simon-

closed 2015-09-10 by sim-

vishnubob, 2015-09-10:
Hi Simon, I forgot to say thank you!  This worked perfectly, and I really appreciate the detailed feedback.  I look forward to using your firmware in on-going and upcoming projects.