sim-/tgy #87: Dys SN20A with RC_PULS_REVERSE=1

fbyland, 2015-08-28:
The fully available power is not correctly scaled over the whole pulse spectrum. If I set the STOP_RC_PULS to 1000 and the FULL_RC_PULS to 2000. The maximums are already achieved around 1950 and 1050 instead of 2000 and 1000. There must be some scaling error.

sim-, 2015-09-10:
Hello! I just tested this and didn't get the same result. Are you sure you're using a version since 02bd8e4ca? I guess that wasn't released until yesterday.

sim-, 2015-09-15:
Hello...Please let me know what setup you have that isn't working, since it works for me (tested a few ways).

fbyland, 2015-09-15:
I have the Dys SN20A ESC with the SimoK Firmware that was the newest on the Git when I posted this Issue.

brescianini, 2015-09-16:
Hi Simon, I did some tests today with the newest version of the firmware. The problem seems to only appear if one uses rcp_alias ("oneshot125").

Settings: 
COMP_PWM = 1
MOTOR_BRAKE = 1
RC_PULSE_REVERSE = 1
STOP_RC_PULSE = 1000
FULL_RC_PULSE = 2000
RCP_DEADBAND = 20
all other settings are their default values

With a regular PWM input signal, I got a 100% output duty cycle for inputs of 1000us and 2000us respectively. However, for a oneshot125-type input signal, I obtained the following duty cycles:

input [us]  -> output duty cycle [-]:
191.25 us (equal to 1530us PWM type signal) -> 12.5%
195 us (1560us PWM)-> 18.9% 
198.75 us (1590us PWM) -> 25.7%
202.5 us (1620us PWM) -> 32.5%
...
and it saturated at about 240.625us (1925us PWM) -> 100%

Hope this helps to reconstruct the error.

Another question: How is the output duty cycle at the RCP_DEADBAND value computed? Even for a regular type PWM input signal it doesn't seem to be a linear scaling from 1500us to 2000us.

Dario

sim-, 2015-09-16:
OK. I did check earlier with a home-made servo tester, but ran out of time and had to go to work. The servo tester I made uses a DAC output on an STM32F4 and the drive strength is so weak that the ramp-time is about 2.2µs, and only reaches about 2.8V, combined with the ATmega8A logic transition level, makes it offset quite a bit. Compensating for that, it looked right. Maybe there is a similar issue with your source?

Note also that I found that the current baseflight code seems to be not quite 1/8 for some reason. Did you check on an oscilloscope? Which source are you using? (I can try to match.)

BTW, RCP_DEADBAND is just added to both sides of MID_RC_PULS as a deadband. So, 50µs there means it won't start until 1460±50µs, and it should scale properly to 1/8 pulse lengths.

brescianini, 2015-09-18:
Thanks for your answer. I only checked the source signal while it wasn't connected with an ESC and it looked fine there. But I'll also check the signal while it is actually connected to an ESC to see if I have a similar issue.

I'll keep you posted.

brescianini, 2015-09-22:
I ran some more tests and I really think there is some weird scaling happening with the "oneshot125" mode. Just to be sure, I used a signal generator this time (t90-t10 rise time is 20ns) but the results were consistent with what I measured last time.

When using a regular PWM input signal, everything works fine and I get a 100% output duty cycle at STOP_RC_PULS and FULL_RC_PULS respectively (although the duty cycle at "MID_RC_PULS +- RCP_DEADBAND" is still magic to me).

When using a "oneshot125"-type input signal, the motor starts to spin at 1/8th of "MID_RC_PULS +- RCP_DEADBAND" as expected. However, a 100% output duty cycle is obtained before the input signal reaches reaches 1/8th of STOP_RC_PULS or FULL_RC_PULS respectively. With the settings
STOP_RC_PULSE = 1000,
FULL_RC_PULSE = 2000,
RCP_DEADBAND = 20,
a 100% output duty cycle is achieved at around 134.375 (1075us PWM) and 240.625us (1925us PWM), respectively.

maxikrie, 2015-11-02:
Hi Simon, 

I am also seeing this issue. I am expecting that a PWM command of 125 us maps to 0 % and 250 us maps to 100 % of the applied voltage. However, I am measuring what you see in the figure as blue dots (the green line being a fitting through the measurements, and the red line what I expect). 

[image]

I am using  a dys ESC in oneshot mode with the following custom settings 

;**********************
; Custom settings     *
;**********************
.equ COMP_PWM = 1 ; During PWM off, switch high side on (unsafe on some boards!)

.equ MOTOR_ADVANCE	= 8 ; Degrees of timing advance (0 - 30, 30 meaning no delay)

.equ MOTOR_BRAKE = 1 ; Enable brake during neutral/idle ("motor drag" brake)

.equ RPM_FB = 1 ; Output sync pulses on MOSI (no flags), for RPM feedback

.equ RC_PULS_REVERSE = 0 ; Enable RC-car style forward/reverse throttle

.equ STOP_RC_PULS = 1000 ; Stop motor at or below this pulse leng
.equ FULL_RC_PULS = 2000 ; Full speed at or above this pulse length

.equ RCP_DEADBAND = 20 ; Do not start until this much above or below neutral

.equ BRAKE_SPEED = 6 ; Speed to reach MAX_POWER, 0 (slowest) - 8 (fastest)

Thanks,
Max