sim-/tgy #89: XXD 30A ESC Callibration Issue?

nicefellow1234, 2015-09-19:
Hi,

I have completed building my F450 Quad with following setup as :
1) XXD A2212 1000 KV Motors
2) XXD 30A ESC's
3) 2650mAH 3S Battery
4) 450mm Frame
5) KK 2.15 Flight Controller
6) Dynam 4 CH Transmitter With 7 CH Receiver
So I am having issue in calibration of ESC's...I can hear the 2nd beeps and also beeps are of differrent sounds..Also I haven't flashed the ESC's...These are running on stock firmware...!
My Motors are turning on different times when I increase throttle..Hence the Quad flips from one side to another..Is not really flying very straight from the ground..!
So any help would be appreciated on this issue please..!

Regards,
Umair Shah Yousafzai

halki, 2016-02-22:
i have same proplem.Did you solve this proplem?

kh4, 2016-02-27:
If the motors behave good with throttle passthru it is obvious that esc calibration is fine and thus this issue is in totally wrong place please close it.  The problem is in your FC board settings/connections/orientation.