sim-/tgy #91: Red Brick 70A Changed Pins

ahalekelly, 2015-09-23:
Hello, I just ordered some new Red Brick 70A speed controllers. They changed the pinout before so that it matched the Red Brick 50A, but it appears they changed it again and now neither work. It appears that at the minimum, the FETs are on different pins because there is no beep when I plug the battery in. Here are some photos of the board, let me know if you want me to probe some pins or test out a new version of the configuration file. I was hoping to use these ESCs at a Maker Faire this Saturday, so it would be great if we could get it working by then. Thank you!
[image]
[image]

sim-, 2015-09-23:
Hello!

Not sure why those factories like to change the pinout so often, but anyway, it's difficult to see more than a couple of traces from the pictures, so I would suggest following https://github.com/sim-/tgy/wiki/Identifying-ESC-pin-configuration to see if you can map out the pin configuration. It's quite likely that one of the other higher current board configurations will work, so probably we just need to compare what you find with some of those .inc files and see if the configuration matches. It's likely something like rb50a, birdie70a, or hk200a. If you have a current-limited supply, you could also just try those, and see if one works. Some may almost work but not quite spin the motor because of wrong sense lines.

Cheers

ahalekelly, 2015-09-24:
Alright, I just spent a couple hours probing the board and this is what I've got. It doesn't follow any rhyme or reason, or match any of the previous ESCs that I can see. I'll try to flash it later tonight.

Assuming phase A is the side with the negative battery connection:

 BnFET PB0

CpFET PC3
CnFET PC4
BpFET PC5

ApFET PD4
AnFET PD5

mux_a ADC6
mux_b ADC0
mux_c ADC7
c_comp AIN0 (neutral)

rcp_in PD2

ahalekelly, 2015-09-24:
I also tried flashing a HKMacro.hex file that I found on a forum, and it did drive the motor, but only supports going in one direction. I can't seem to find any information on it, is it an OEM firmware that someone pulled off their ESC? If we could find the source we could pull the pin assignments.

sim-, 2015-09-25:
That's probably George's code. He has something written in C that he uses for the F5B stuff.

Seems maybe a board I don't have a pinout for yet, since nothing has BpFET on PC5 and BnFET on PB0. Are those drivers inverted? Not finding the datasheet easily. Maybe I'll email George. :)

Meanwhile, you could try to make an .inc by copying or modifying the existing "bs.inc", which is close.

ahalekelly, 2015-09-25:
The drivers are generic 5016 chips, I found datasheets from [TI](http://www.ti.com/lit/ds/symlink/lm5106.pdf) and [ON Semi](http://www.onsemi.com/pub_link/Collateral/NCP5106-D.PDF). Neither of them mention inversion, so can I assume it's not inverted? I already made a fork of this repo and [modified the old rb70a file](https://github.com/ahalekelly/tgy/blob/master/rb70a_new.inc) if you want to take a look.

ahalekelly, 2015-09-25:
It does seem to be George Shering who originally posted that code.

And last time I was using a current limited power supply while messing with firmware versions. The only time I ever hit the 0.5A limit was with George's code, running a motor :p I just plugged the ESC into my 9V battery to program it, my cheap semi-current limited supply, and 5 seconds later one of the FETs let out the magic smoke. It's rather strange, I'll try again after I solder up my second (and last) red brick.

sim-, 2015-09-26:
Ahh, I was looking at George's post saying "6105 FET drivers". Yes, those look just like simple drivers without inversion (logic high means ON for both pins).

Hmm, magic smoke from a 9V battery? That's odd. Maybe try your .inc with CHECK_HARDWARE=1, which should not proceed if it can't verify that all drive and feedback is working properly. Well, it will try to report issues if they are non-fatal. You could also wire an LED-resistor to some pin and define it so that it can report that way.

Anyway, what should we call this? I hate calling things "new" because inevitably they won't be as soon as the next one comes out. I wonder if it matches the Mystery 200A (bs200a?), or if it's really more Red Brick-ish. So many variants of the same boards >=40A, it's difficult to come up with a good scheme.

George replied with his pin configuration and it seems to match what you have, but with A and C flipped. This is the same other than the default rotation direction. Can you do a favour and see if your .inc also matches the stock rotation direction without changing the motor wires? I like to make it match the stock firmware where possible. (If it's backwards, we can just swap everything A and C.)

Thanks!

ahalekelly, 2015-09-30:
I soldered up the second ESC tonight, flashed it and it works great! I was using an arduino to test the ESCs and wasn't able to get it working with the stock firmware before I flashed it, so I don't know if this matches the stock direction. I got someone else to test as well, and they confirmed it works on both the Red Brick 70A and the new version of the Hobbyking SS 190-200A. I'll add this to the list of ESCs in tgy.asm, and then I can put in a pull request.

The magic smoke happened when it was running one of the random .inc files that I was trying before on the current limited power supply. Somehow it didn't trigger the current limit then, but blew out when I plugged it into the 9V. Oh well.

As for what to call it, I agree that "new" is a bad name. We could call it by the date it was first seen, or the number of different revisions that have been seen (from what I've read online this is the third version of the pinout, the second one worked with the rb50a configuration), or by the revision number on the PCB (4P0529A), which might change in the future without the pinout changing. This is also the first version I've heard of to use a blue PCB, but that has the same issue. I think the most foolproof solution is to put a description of the ESC with all of these attributes, as well as pictures of the PCB, at the top of the .inc file.

Thanks for all the help! I have a couple more questions about the inner workings of the firmware if you don't mind, where would be the best place to ask them? This has been taking off in the robotics world and the requirements can be a little different than in a multirotor.

ahalekelly, 2015-10-02:
So what do you want to name the new file?