sim-/tgy #95: Variable Phase Advance

ahalekelly, 2015-10-17:
Sorry if you're already doing this, the assembly is a little dense for me. With a static phase advance value, you are optimizing the motor efficiency at a single speed, while many other brushless speed controllers change the amount of motor phase advance linearly in proportion to the motor speed to be as efficient as possible at all speeds. In that case you set the motor advance value for a specific eRPM value. Am I just missing the spot in the code where you are doing this? If so, what eRPM value is the MOTOR_ADVANCE variable for?

ahalekelly, 2015-10-17:
As I think more about this, it seems that a value for phase advance in terms of time is the exact same thing. For example, a 1 microsecond advance is 0.000006 degrees/rpm. It seems that TIMING_OFFSET does just this, correct?

sim-, 2017-02-16:
Hello! Yes, but finding it automatically is currently not supported. I'm not sure how it might be best to find this automatically. Meanwhile, Owen has a calculator here: http://owenduffy.net/rc/tools/bldca.htm