sim-/tgy #100: Underwater thruster gets underpowered when inserted in water

smosh, 2015-10-27:
I'm part of a team/club attempting to build an autonomous submarine, and we want to use these [videoray brushless thrusters/motors](http://www.videoray.com/images/specsheets/2013/Brushless_Thrusters.pdf). But we ran into a particular issue in that they work fin in air, but the second you put the propeller in water, the motor controller seems to limit the power the thruster receives drastically. We're kinda lost as to why this is happening, and we're coming here for any help we can get with understanding this issue.

[here is a video demonstrating our problem](https://youtu.be/xdjh5uBd68k)

Our circuit setup is shown below, whereby we are using a HK 20A motor controller flashed with the bs_nfet.inc firmare from this github project, we're guessing that this is part of the problem, but it's a very weak guess.

[image]

Is there any kind of protective function that bs_nfet.inc exhibits when a propeller is subject to a much denser medium than air? If one looks closely at the provided video, you can see that there is a discrete step where the motor controller seems to command the thruster to spin at a lower speed.

Also, we know the propeller is not slipping. We've eliminated that as a potential cause of the problem.

# More detailed lab data
If it helps, there's some more detail I can offer about the motor's electrical characteristics surrounding this problem:

The picture below shows the thruster spinning in air at low speed. One of the motor phases is probed, and is represented by the pink waveform. One interesting thing to note is that the underlying PWM is not completely square when the thruster is in air. (You can kinda see this in the video too).

[image]

The picture below is of the thruster spinning at about the same speed, but in the water. This time, the underlying PWM signal is completely square. We're not sure why, or if this is even significant. 

[image]

kh4, 2015-10-27:
Well running the thruster in air makes very little sense in the first place. Instead keep it in water and play with the settings a bit. I fear it may need a bit special ESC due to low RPM (and thus very low level of back EMF).

sim-, 2015-10-28:
Hello! Just quickly, if you start the motor under water, what happens? I'm guessing that it will start but not ever reach full duty? The likely cause is that the RPM is low enough that it is being limited by the timing-based duty limiting, which is there to reduce the chances of misalignment at low speeds.

You can adjust or bypass this, but it would seem to be unusually slow for reasonable efficiency. I would expect gearing (electrical or mechanical) to make more sense in that case, unless you are just testing at particularly low input voltage (under the rating). Beware high currents.

In any event, TIMING_RANGE3 is the parameter that would need adjusting, but I can't remember the details on how far that can be adjusted. Try doubling the value (which is an interval) it to halve the minimum speed where limiting duty starts, for example.

smosh, 2015-10-28:
sim-, you are right: if we start the motor underwater, it never reaches full duty cycle. Prior to this video, we did try running the thruster underwater. The knob/potentiometer used to control the speed seems to have it's range reduced to about 1/8 the total range when the thruster is in the water. Control of the motor feels linear within that range, but commands outside of that range are ignored. Thanks for point out TIMING_RANGE3. We'll try changing that parameter and seeing what happens.

kh4: I didn't know we would see a difference among the three phases. I'll try probing the other phases too to see what we get.

kh4, 2015-10-28:
Well all phases should be identical (just with 120 degree phase offset), it is just easier to figure out with them all visible

smosh, 2015-10-29:
I can confirm that changing TIMING_RANGE3 from 0x2000 to 0x4000 and reflashing the ESC. The thruster seems to have it's full duty cycle in water now. Thanks!

smosh, 2015-10-30:
So, we observed another thing we while testing with TIMING_RANGE3 = 0x2000, the current the thruster draws is no longer periodic. For instance here is a scope-shot of the thruster (VR4) being driven at low speed in water:

[image]

The current profile is pretty much sinusoidal.

On the other hand, when we drive the thruster at max speed, we get a more irregular waveform. Here is a scope-shot of the VR4 being driven at max speed in water:

[image]

I'm guessing this is something abnormal and/or inefficient. Is this something that can be fixed by changing some more parameters in the firmware, or would we be better off trying a different motor controller?