sim-/tgy #110: no need for pulldown resistors.

rewolff, 2016-02-13:
> If the hardware does not have any low pass filter or pull-down on the PWM input, it may easily float high by the time the boot loader checks it, which can prevent normal startup if another input (eg: I2C) is expected to be used. There should be pull-down or load present on the PWM input to prevent this if BOOTRST is enabled.
If, say 5 clock cycles before checking you make the PWM input pin an "output 0" for ONE clock cycle, it will be low when you check it if unconnected. The AVR can unofficially handle the 62ns of "too high" current through the output transistor (about 160mA). 

Would you feel bad about having 100nF on an output pin? At 100nF there would be about 6% less power in the output transistor during that first clock cycle after trying to discharge that capacitor. And the capacitor would dump about 25 times more energy in the output capacitor if left to discharge completely (1.25uJ for the cap versus 0.05uJ for the output-0-while-shorted-to-VCC for one cycle.

sim-, 2016-02-13:
OK, but, is there a particular use case that is broken without this?

rewolff, 2016-02-14:
The directions state to "add hardware" (a pulldown) when I think the issue can be solved in software. This could be a barrier for people who are less proficient with the soldering iron.... 
The pulldown is easy enough for you and  me, but I notice many others are hesitant to grab the soldering iron. (I gave my girlfriend a soldering iron when she graduated: "you can't be an electrical engineer if you don't own a soldering iron". Recently, twentyfive years later it has seen its first use.... Geesh!)

sim-, 2017-01-13:
Hehe :)

Forgot to follow up on this.. The only hardware I've helped design has the pull-down, or has only I2C, so this is not a problem in practice as far as I know. If you happen to have an ESC you've gone to the effort to make in the first place, you can simply disable anything but the input that is desired in software. I don't think that intentionally draining the pin in order to stop it from floating high is a good idea, since it has to not do this on ESCs which have no or low resistance. Adding any capacitance starts to skew timing which has become more important with the latest trend of very short pulse lengths.

closed 2017-01-13 by sim-