sim-/tgy #111: BS 40A - Motors "hammering"

ashankaei, 2016-04-04:
Hi,

I tried to flash the Blue Series 40A, with bs40 and bsnfet and they both have an issue. When I turn on the motor(s) they do some sort of hammering as if they want to go CW and CCW in different cycles.
I checked my wiring and used different ESCs, Motors and Batteries they look O.K.

My guess is that, the phases are not following each other but I'm not 100% sure! Did this problem occurred to you before?

sim-, 2017-01-13:
Hello! That sounds mostly like perhaps the wrong pinout. 40A is likely not bs_nfet but is likely bs. Did you get this working in the meantime?

kasparsd, 2017-03-24:
> When I turn on the motor(s) they do some sort of hammering as if they want to go CW and CCW in different cycles.

Had the exact same issue with `bs40a` today. With `bs_nfet` I could hear the FETs switching but the motor was not running at all.

> That sounds mostly like perhaps the wrong pinout

@sim- Are you referring to the `XnFET` and  `XpFET` pins?

kasparsd, 2017-03-24:
@ashankaei @sim- Turns out the pins are different. Here is a fix #128