sim-/tgy #115: All n type bridge vs n and p type.

alnaufra, 2016-06-06:
Hi,
You say that cheap esc have p type fets in the hight side and n type on the low side, and that an improvemnt to that is a all n type bridge.
I don't see why, could you explain me why all n type mosfets is better?
From my point of view(I'm doing an esc with arduino) it's easier the first way and i can't see the down side.
Thanks in advance.

orgua, 2016-06-06:
try to find a 100A p-Fet for under 30$ - if there is one at all. n-fet supports much more current than p and costs <1$ for 100A, but requires a special driver if you want to use it on the high side.

alnaufra, 2016-06-06:
OK! so is just price based for hight current. Perfect thanks a lot for the fast reply!

closed 2017-01-13 by sim-