sim-/tgy #116: Asymmetric reversal response issue?

pumaking, 2016-06-24:
I'm seeing a peculiar issue where driving a BlueRobotics T200 in the reverse direction and then quickly driving it in the forward direction causes it to appear to wait for the motor (propeller) to stop spinning before driving it forwards.

However, driving forwards and quickly reversing, works as intended; the motor reverses direction immediately.

Does anyone know if this is something the firmware does intentionally or some kind of error? We are using ZTW Spider ESCs, specifically:
http://www.hobbyking.com/hobbyking/store/__81206__ZTW_Spider_Series_30A_Small_OPTO_Multi_Rotor_ESC_2_6S_SimonK_Firmware_.html

Thanks!

sim-, 2017-01-13:
Hmm. I have no idea what firmware they are using on those. There is nothing I can think of that would (intentionally) result in that. Did you ever figure out what was going on here?

It's odd that they say "SimonK" but have a programming manual that specifies stick programming of motor timing, etc. Sounds like they aren't using my code, or maybe the manual is wrong.

Anyway, in order to make it reversible, you probably had to flash it anyway, right?

The only thing that comes to mind is that ZTW typically goes with the Mystery/BlueSeries (usually bs_nfet) schematic, which often carries with it some filtering capacitors that make timing not consistent across the RPM range and make starting somewhat worse. However, I would expect it to be consistent regardless of direction.

farayman3, 2020-03-09:
I @sim- I seem to be having the same problem. I though it was random but after reading pumaking's comment it seems indeed to only happen (sometimes) when going from reverse to forward. If I go really quickly from reverse to forward, occasionally it seems as if the brake function is not working. The motor seems to wait until it stops spinning by itself and then suddenly goes into the forward drive. I am using a Hobyking F60A. SimonK stock settings but RC_PULS_REVERSE = 1, MOTOR_BRAKE = 1 and COMP_PWM = 1. If you need more information / a video showing the effect, let me know.