sim-/tgy #119: PB6 & PB7 (OSC) use for XnFET & XpFET

Jonny89, 2016-10-19:
Hello Guys,

i want to build my own ESC with Simonk. Now i'am working on the layout around the Atmega8 and want to change some pinconfigurations for a better way out.

So is it possible to use PB6 and PB7 (normally used for the Oscillator) for the signals XnFET and XpFET?
The ESC will run without external oscillator.

I hope to get a postive reply ;)
Best regards Jonny

IHaveADrone, 2016-10-20:
IIRC, you need the precision of an external crystal

sim-, 2016-10-20:
Sure, but as noted, the ATmega8 internal RC oscillator is kind of awful. It is very temperature-sensitive. Otherwise, it should work without issue (the original tgy board, for example, is common and does not have an oscillator).

closed 2017-01-13 by sim-