sim-/tgy #133: e-skateboard brakes

youbobcat, 2017-10-15:
Trying to replicated https://endless-sphere.com/forums/viewtopic.php?f=35&t=74880 braking.

currently hacking:

run_reverse:
		rcall	wait_for_low
		nFET_brake_on i_temp1
		sync_on
		rcall	wait_for_high
		nFET_brake_off i_temp1
		rcall	wait_for_low
		nFET_brake_on i_temp1
		rcall	wait_for_high
		nFET_brake_off i_temp1
		sync_off
		rcall	wait_for_low
		nFET_brake_on i_temp1
		rcall	wait_for_high
		nFET_brake_off i_temp1

	.macro nFET_brake_on
		AnFET_on
		BnFET_on
		CnFET_on
	.endmacro
	
	.macro nFET_brake_off
		in	@0, AnFET_port
		.if (INIT_PB & ((AnFET_port == PORTB) << AnFET)) | (INIT_PC & ((AnFET_port == PORTC) << AnFET)) | (INIT_PD & ((AnFET_port == PORTD) << AnFET))
		cbr	@0, (1<<AnFET)+(1<<BnFET)+(1<<CnFET)
		.else
		sbr	@0, (1<<AnFET)+(1<<BnFET)+(1<<CnFET)
		.endif
		out	AnFET_port, @0
	.endmacro

This is as far as I have gotten any help would be very much appreciated!

tracernz, 2017-10-15:
http://vedder.se/2015/01/vesc-open-source-esc/ might be of interest to you if you haven't seen it.

youbobcat, 2017-10-17:
Thanks I have read about VESC but my aim was to get working a cheap redbrick ESC I own. I have everything but the dynamic throttle controlled brakes working.

Tomonickb, 2020-06-24:
Did you made any progress? I am now dealing with the same thing :(, recently bought RED BRICK 100A and find out that it is braking too hard.