sim-/tgy #141: Yellow ESC

diego0815, 2021-01-14:
Has anyone been successful flashing those yellow Chinese ESCs? 
I stripped one, see board shots below:
[image]

[image]

sim-, 2021-01-14:
Hello! At a quick glance, I'd guess they're tgy.hex-compatible, as there is no external oscillator and two chains of transistors for discrete FET driving. I'd say try flashing one with a current-limited supply and see if it works or not.

diego0815, 2021-01-14:
Thank you very much, will try as soon as I'm feeling courageous enough to solder to the pins and to unpack my avrdragon... It's a long time I didn't :-)

a146860, 2021-01-14:
Compatible, please contact me to buy 14686093@qq.com

robante15, 2024-10-20:
Were you able to perform this procedure?