sim-/tgy #145: Firmware for this yellow generic cheap ESC

batata004, 2021-08-21:
This is one of the most sold ESCs in Brazil and I despite being very common, I cant find any reference pointing me to the right firmware to flash.  Would you mind taking a look at the pictures below and guessing which firmware should I try to flash using BLHELISUITE? 

Thank you and I appreciate the job you are doing for the community keeping this repo alive!

[image]
[image]

StoicaMad, 2022-05-23:
try with <<tgy_8mhz.hex >>

dinho-limno, 2023-06-21:
Fala Brother, vc conseguiu atualizar o firmware desse esc para o blheli? Parece que esse amarelinho usa o SimonK e quero muito usá-lo para reverter o sinal do motor.

robante15, 2024-10-20:
Were you able to perform this procedure?